redbear/Duo #2: Provide option on Arduino to upload through RBLink

rei-vilo, 2016-02-04:
Provide option on Arduino to compile and upload through RBLink, as the RBDuo appears as a mass storage device to drag-and-drop the executable into, _à la_ mbed.

Cheong2K, 2016-03-10:
We are working on using OpenOCD to burn system firmwares and user part using the RBLink from Arduino IDE.

Cheong2K, 2016-03-29:
@rei-vilo, would you mind to help to test uploading sketches using the RBLink (with Board Package 0.2.7)?
Ref.:
https://github.com/redbear/STM32-Arduino

rei-vilo, 2016-03-29:
Sure, where can I find the procedure? It doesn't seem to have been published yet on the linked page.

rei-vilo, 2016-04-01:
Ok, `RedBear Duo (RBLink USB Port)` now appears on the boards after updating to release 0.2.7.

Upload works fine. Console serial is `Serial1`.

When connecting the RBLink, a mass storage device appears names `RBLink` is mounted. Any plans to implement a copy-paste-program like in mbed?

closed 2016-04-01 by rei-vilo