redbear/Duo #8: When flashing BLE scan example Duo can't connect to Particle Cloud anymore

plaetzchen, 2016-02-17:
Steps to reproduce:

1. Connect Duo to Particle Cloud
2. Upload Sketch from https://github.com/redbear/STM32-Arduino/tree/master/arduino/libraries/RedBear_Duo/examples/03.BLE/BLE_Scan with Arduino Library imported as #include "Arduino/Arduino.h"
3. Flash firmware. Scan works normally
4. Try to change code and flash again. Duo is not connected anymore

To reset a flashing with dfu-util is needed.

Cheong2K, 2016-02-17:
The example should be updated with more description.

It uses manual mode, so it will not connect to the network:
SYSTEM_MODE(MANUAL); //do not connect to cloud

Please change it to:
SYSTEM_MODE(AUTOMATIC); //connect to cloud

plaetzchen, 2016-02-18:
Ah OK, I see. I will try that later today.

plaetzchen, 2016-02-18:
OK I got that running. Is there any documentation about the ble. methods you are using? I can't find anything about them at Arduino.cc

sthanh, 2016-07-08:
Any more progress on BLE scanning development? I'm looking to use Particle Cloud for scanning Eddystone and the example doesn't have clear instructions on how to scan and report the found beacons to the cloud.

Thanks

XuGuohui, 2016-07-13:
@plaetzchen , Please refer to the programming reference munual here:
https://github.com/redbear/Duo/blob/master/docs/programming_reference_manual.md#reference

closed 2016-07-13 by XuGuohui