redbear/Duo #10: DFU install on OS/X with brew

paultanner, 2016-03-05:
I do not use brew very often so not sure what to do where this happens:

````
Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink bin/dfu-prefix
/usr/local/bin is not writable.

You can try again using:
  brew link dfu-util
````
So I thought ok use sudo brew link  dfu-util but I get
````
Error: Cowardly refusing to `sudo brew link`
You can use brew with sudo, but only if the brew executable is owned by root.
However, this is both not recommended and completely unsupported so do so at
your own risk.
````
So I think the instructions need to be clearer

Cheong2K, 2016-03-05:
Which OSX version are you using?
Please try this one at the moment.
https://github.com/redbear/STM32-Arduino/raw/master/arduino/tools/fw_loader/macosx/dfu-util

paultanner, 2016-03-05:
Sorry, I should have said that I am using OS/X 10.11.3 (El Capitan)
The first link you sent is not an OS/X package.  If it's a binary I'm not sure how to *safely* install it.
I think I'll wait till I really need it then download the source and build.
Regards, Paul

Cheong2K, 2016-03-10:
Hi, does it work for you now?

paultanner, 2016-03-14:
Sorry, I have not had time to do a build.  As you know, this is not 
always a quick thing

Regards, Paul

JeroenConnectivity, 2016-06-02:
http://dfu-util.sourceforge.net/build.html

closed 2017-03-04 by Cheong2K