redbear/Duo #14: Stuck in Green blinking loop

JeroenConnectivity, 2016-05-31:
Hello, 

After flashing FW with particle.io my redbearDuo is stuck in a blinking Green light cycle. I cannot change to FW via serial because there seems to be no valid DFU suffix signature:

Filter on vendor = 0x2b04 product = 0xd058
Opening DFU USB device... ID 2b04:d058
Run-time device DFU version 011a
Found DFU: [2b04:d058] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash   /0x08000000/01*016Ka,02*016Kg,01*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
No valid DFU suffix signature
Warning: File has no DFU suffix
DfuSe interface name: "Internal Flash   "

Any ideas how to solve this isue?

JeroenConnectivity, 2016-06-01:
I did manage to flash bootloader via Arduino IDE, but the connection with Particle Cloud IDE is still broken. It seems to be stuck in Safe-Mode. Any Help?

XuGuohui, 2016-06-02:
Hi @JeroenConnectivity ,After you flashing bootloader, what have you done to make your Duo connect to cloud? You said it seams in safe mode, did the RGB breathing magenta?

JeroenConnectivity, 2016-06-02:
@XuGuohui Thanks for the reply! I tried all kinds of SYSTEM_MODEs. It seems to run Arduino code when I put it on MANUAL, but the other modes (which connect to Cloud) makes the Duo go into the following state: rapid blinking blue, mixed with rapid blinking yellow and that in than blinks at slower speed. This cycle repeats. Seems like the Particle section of the Duo stuck in some sort of Safe-Mode

XuGuohui, 2016-06-02:
It may be blinking cyan rather than blue, right? In that case, you should follow this guide to provision your Duo:

https://github.com/redbear/Duo/blob/master/docs/provisioning_guide.md

JeroenConnectivity, 2016-06-02:
Whats the content of the main.js file you're referring to?

XuGuohui, 2016-06-02:
You need to download or clone this repository:

https://github.com/redbear/device-provisioning-helper

JeroenConnectivity, 2016-06-02:
Worked through the process and one thing was a bit strange. When I try to load the DCT I get the response: "(C) 2005-2008 by Weston Schmidt, Harald Welte and OpenMoko Inc.
(C) 2010-2011 Tormod Volden (DfuSe support)
This program is Free Software and has ABSOLUTELY NO WARRANTY

dfu-util does currently only support DFU version 1.0

Filter on vendor = 0x2b04 product = 0xd058
Cannot open device 
Opening DFU USB device... "

Device-provisioning-helper is working though. Gave me the proper response after execution.

JeroenConnectivity, 2016-06-02:
Sorry forgot sudo:P Now the response is: "
Filter on vendor = 0x2b04 product = 0xd058
Opening DFU USB device... ID 2b04:d058
Run-time device DFU version 011a
Found DFU: [2b04:d058] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash   /0x08000000/01*016Ka,02*016Kg,01*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuDNLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
No valid DFU suffix signature
Warning: File has no DFU suffix
DfuSe interface name: "Internal Flash   "
 "

XuGuohui, 2016-06-02:
Did you put your Duo in DFU mode before loading the DCT image?

JeroenConnectivity, 2016-06-02:
Yes

XuGuohui, 2016-06-02:
What is the command you run the dfu-util?

JeroenConnectivity, 2016-06-02:
dfu-util -d 2b04:d058 -a 0 -s 0x08004000 -D duo-dct-r1.bin

JeroenConnectivity, 2016-06-02:
The execution of dfu-util just stops after Interal Flash message while it should start Downloading process (which I've seen happening during bootloader flash in Arduino IDE)

XuGuohui, 2016-06-02:
Are you using the Mac OSX or Linux? If so, try run the dfu command with `sudo`

JeroenConnectivity, 2016-06-02:
Did run with sudo. Using Linux

XuGuohui, 2016-06-02:
Please run command `dfu-util -l` to check is there is any device in DFU mode.

JeroenConnectivity, 2016-06-02:
dfu-util does currently only support DFU version 1.0

Found DFU: [2b04:d058] devnum=0, cfg=1, intf=0, alt=0, name="UNDEFINED"
Found DFU: [2b04:d058] devnum=0, cfg=1, intf=0, alt=1, name="UNDEFINED"
Found DFU: [2b04:d058] devnum=0, cfg=1, intf=0, alt=2, name="UNDEFINED"

XuGuohui, 2016-06-02:
Did you install dfu-util by following this guide:

https://github.com/redbear/Duo/blob/master/docs/dfu.md

JeroenConnectivity, 2016-06-02:
I used sudo apt-get install dfu-util

XuGuohui, 2016-06-02:
It's weird! Can you load system firmware using dfu-util?

JeroenConnectivity, 2016-06-02:
Nope. It sure is;P  I'll be oke with a workaround (bypass dfu-util) to factory reset the FW with the Arduino IDE so I can use the particle IDE again. Is this possible?

JeroenConnectivity, 2016-06-02:
Arduino Consol: 

Opening DFU capable USB device...
ID 2b04:d058
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Serial Flash   "
Downloading to address = 0x00140000, size = 6208

Download	[                         ]   0%            0 bytes
Download	[================         ]  65%         4096 bytes
Download	[=========================] 100%         6208 bytes
Download done.
File downloaded successfully

Linux Terminal:

Opening DFU USB device... ID 2b04:d058
Run-time device DFU version 011a
Found DFU: [2b04:d058] devnum=0, cfg=1, intf=0, alt=0, name="@Internal Flash   /0x08000000/01*016Ka,02*016Kg,01*016Kg,01*064Kg,07*128Kg"
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuDNLOAD-IDLE, status = 0
aborting previous incomplete transfer
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
No valid DFU suffix signature
Warning: File has no DFU suffix
DfuSe interface name: "Internal Flash   "

XuGuohui, 2016-06-02:
The Arduino burn bootloader option does not reset the DCT. If you didn't ever do something to DCT, it should be OK and you need not to reset it and just moving on.

XuGuohui, 2016-06-02:
By comparing the message between Arduino and terminal, when running dfu-util from terminal, it can not identify the interface name `DfuSe interface name: "Serial Flash ", which may terminate the progress.

JeroenConnectivity, 2016-06-02:
Oke fixed the dfu-util by updating to 0.8. In apt-get 0.5 gets installed. I ran following commands:

git clone git://git.code.sf.net/p/dfu-util/dfu-util
cd dfu-util
./autogen.sh
./configure  # on most systems
make
sudo make install

JeroenConnectivity, 2016-06-02:
I followed the Provisioning Guide again, but i'm unable to claim my device. I deleted the duo from the particle IDE. Should I be able to reclaim the device after the provisioning process?

XuGuohui, 2016-06-02:
Glad you fix the problem. Maybe I can add this to the dfu-util installation guide. :+1: 
Yes, definitely you can reclaim it, but only when it is online.

JeroenConnectivity, 2016-06-02:
Got it online by uploading Arduino Sketch with SYSTEM_MODE(AUTOMATIC) defined. Then it entered listening mode which allowed me to reconnect to cloud and made it possible to reclame afterwards! **Thanks for the help** and good idea to update installation guide!

closed 2016-06-02 by JeroenConnectivity