redbear/Duo #18: Losing Wifi Connection

sthanh, 2016-07-12:
Hi there,
I have issue of intermittently losing wifi connection. It connect to the Particle cloud and publish the data just fine but after about 20-30 minutes, it loses connection and the LED flashing green. A hard reset sometime put the wifi back online and sometime, it stuck at flashing green. Any ideas what's going on? I'm using the latest firmware (0.2.4).

Thanks,
Shawn

XuGuohui, 2016-07-13:
Hi @sthanh ,

This is probably caused by an unstable cloud connection or Wi-Fi connection. Anyone might suffer from this situation.

If the Duo is flashing green, it means that the Duo is trying to connect to your AP (router).
If the Duo is flashing cyan, it means that the Duo is trying to connect to the cloud.

Cheers!

sthanh, 2016-07-13:
@XuGuohui Thanks for the response. I ran next to the Pi 3 and the Pi3 was running just fine without losing wifi connection. A few people also report similar situation over at the Particle Community. 
[https://community.particle.io/t/photon-loses-connectivity-very-often-only-reconnects-on-hard-reset/12680/10](https://community.particle.io/t/photon-loses-connectivity-very-often-only-reconnects-on-hard-reset/12680/10)

[https://community.particle.io/t/photon-losing-wifi/12321](https://community.particle.io/t/photon-losing-wifi/12321)

suranjandot, 2016-10-29:
hi!

Even I have the same problem. The Duos remain connected for hours, and then lose the connection and go green. Other devices on the network are still able to connect. No network settings have been changed, and wifi credentials are the same.

Restarting the duo (powering off and on) makes it connect and breathe cyan.

Why is this happening? And what can I do to solve it?

My duos are installed in a place where they cant be manually monitored or restarted. They must stay running and connected for weeks as long as the WiFi doesn't go off. 

Thanks!

suranjandot, 2016-10-29:
Edit: I have 20 rebear duos running the latest firmware v0.2.4

XuGuohui, 2016-10-31:
@suranjandot Will the Duo automatically connect to cloud again if you don't power it cycle?

billskeen68, 2016-11-02:
@XuGuohui I still have this problem as well (See RedBear Duo Fourm http://discuss.redbear.cc/t/tcpclient-write-5-second-timeout/710/18). 

For some time I thought it was due to the TCPClient failing. I now believe that it is the WiFi randomly disconnecting and not reconnecting. I too have other devices that are not losing WiFi connections to the same AP; three raspberry Pi 2 and one Pi 3, two laptops running windows 10, one laptop running Linux and a Samsung S6. Only the seven Duos drop connections randomly. The only way to reconnect is to power cycle. A system reset will not work. It is like the Broadcom chip does not reset properly.

All but two Duos have an external antenna. Checking with Android WiFi Analyzer at each Duo shows -60db or better RSSI to AP. The Duos are powered with either a 3.3v or 5.0v AC to DC power supply having 500ma or greater output or using a CanaKit 2.5a, 5v external power supply through Duo micro USB serial port. The only other device is a .96" OLED display that consumes .08w max.

Also running BLE with 4 devices connected and receiving notifications without any random BLE disconnects.

Since I connect to WiFi in setup(), how should I code to automatically reconnect the WiFi?

robdobsn, 2017-03-30:
I have a currently open request at https://community.particle.io/t/active-tcpserver-client-appears-to-crash-photon-when-wifi-disconnects/27130
Tlangmo on the particle forum https://community.particle.io/users/tlangmo
has made a suggestion that there is a bug in the TCPServer code and has issued a pull request which still needs verification but might be to blame for some of the problems.

frmdstryr, 2017-05-24:
Also same issue, make this thing pretty much useless when it can't stay connected for more than a few hours.

suranjandot, 2017-05-24:
I have a suggestion. 

WiFi connectivity is handled entirely by the broadcom stack. There seems to be instances even on the Photon when the WiFi module is unable to connect and gets stuck without attempting reconnection.

Use SYSTEM_THREAD(ENABLED), and SYSTEM_MODE(SEMI_AUTOMATIC) if possible in your context. Don't forget to connect to the WiFi and call Particle.connect() on startup.

If it works for you, add a timer, and if WiFi.ready() is false for too long (or better still if Particle.connected() is false for too long) issue a System.reset()

System thread will prevent the application loop and system network processing from blocking each other and the soft reset call should get your duo back online.

For a totally separate yet important reason I think it's better to check Particle.connected() and not Wifi.ready() in duos that are connected to the cloud. Particle keys get renewed from time to time with a fade out of the old ones. If your device was disconnected from the cloud for too long, it may not get the new keys and may not be able to connect to the cloud. This may be important to you if your duo has to run weeks without intervention.

frmdstryr, 2017-05-24:
I'm using SYSTEM_MODE(MANUAL) and using a TCPServer. Will try doing what you suggested and see if it fixes it. Thanks for the prompt reply!

frmdstryr, 2017-11-08:
Neither reconnecting wifi nor a System.reset() works.

robdobsn, 2017-11-10:
I've been using the Particle Firmware v0.7.0-rc.3 with a Photon and, in my tests so far, it is solid. It fixes a number of bugs in the handling of the interface to the WiFi system. RedBear firmware is quite a long way behind the Photon firmware now but the v0.7.0-rc.3 is still a release candidate (as its name implies) so I don't think there's much chance of seeing a RedBear update to this level any time soon unfortunately. I keep intending to have a look at the key changes and see if they can be easily ported across for my own purposes and I would dearly like a working RedBear Duo stack.