redbear/Duo #19: Device unable to start(10)

adithya003, 2016-07-21:
I did factory reset to the device. I released the setup button when the green light was blinking after that i the device is constantly blinking white light for every 5 seconds and i cannot see the device in any port under my laptop.
I tried to install driver again using the DFU mode but still i am getting the same issue how to fix this problem
Is my device broken?? else is their any solution

Thanks in advance

XuGuohui, 2016-07-21:
@adithya003 , can your Duo enter Safe Mode? Please have it a try and then upload a known valid user application via Particle Build or Arduino IDE. 

After that if it works, then the factory reset application may be broken, you have to update it using dfu-util:

https://github.com/redbear/Duo/blob/master/docs/firmware_deployment_guide.md#update-factory-reset-application-fac

adithya003, 2016-07-24:
My device is not being listed in the PORT of the arduino IDE.

adithya003, 2016-07-24:
[image]

This is the error which i am getting 
When i tried to enter into the safemode after 5second the while and blue light are flashing for every 2seconds

XuGuohui, 2016-07-25:
How did you make your Duo enter Safe Mode? It should be:

1. Press both of the button
2. Release only the RESET button
3. Release the SETUP button when you see the RGB blinking magenta 

In Safe Mode, your Duo will try to connect to the Particle cloud, but if you haven't stored any WiFi credentials, it will enter Listen Mode (i.e. RGB blinking blue) first to allow you configure an credential.

Regarding to the driver issue, the driver files may be broken. You can download the origin files here (the .cat and .inf):

https://github.com/redbear/Duo/tree/master/driver/windows

adithya003, 2016-07-25:
[Video Jul 25, 3 20 57 PM.zip](https://github.com/redbear/Duo/files/382316/Video.Jul.25.3.20.57.PM.zip)

This is the result which i am getting when i try to enter into the safe mode.

XuGuohui, 2016-07-26:
I see. Please follow the following link (Back to factory default state) to recover your Duo:

http://discuss.redbear.cc/t/general-recovery-of-the-duo-from-stuck-state/216

adithya003, 2016-07-28:
Under the first step of the factory reset which was mentioned in the given link i downloaded the STM32 ST-LINK util application and followed all the steps when i click connect it says me not detected.