redbear/Duo #21: HomeKit sample code

rajtrada, 2016-09-19:
Hi, 

Could you please provide the sample code for all platform which you shown in video https://www.youtube.com/watch?v=JswRiM87LZc

We have acquired MFI license, I can share details with you if required.

Cheong2K, 2016-09-19:
https://github.com/redbear/WICED-SDK

rajtrada, 2016-09-19:
Thanks for the response,

I have checked that link and couldn't find much information which I am looking for. Could you please specify that example works over BLE or WiFi? 

Basically I am looking for how to control HomeKit enable device from Android.

kalpesh-lakhani, 2016-09-19:
Are you mad or what ?
why do u want to do this things ?
just chill..

if it about earning money then there are so many ways to do.

Cheong2K, 2016-09-19:
I am not sure about that, too. I suggest you ask these questions at Cypress WICED forum, they have experts to help you.

closed 2017-03-04 by Cheong2K