redbear/Duo #31: Duo V1.0 no BLE connection with Firmware 0.3.x

sunnycyk, 2017-06-26:
After upgrade to firmware 0.3.0 or 0.3.1, BLE connection is not found on Listening Mode.

TechplexEngineer, 2017-07-26:
This seems to be expected behavior:
> Since system firmware v0.3.0.
> 
> Before v0.3.0, when the Duo enters the Listening Mode, it will broacast as a BLE peripheral for user to configure WiFi credentials etc.
> 
> Since v0.3.0, user now can optionally enable this feature. It is disabled by default. But you can declare it in your application explicitly by adding BLE_SETUP(DISABLED). To enable this feature, you should declare BLE_SETUP(ENABLED) in your application. 

See: https://github.com/redbear/Duo/blob/master/docs/programming_reference_manual.md#ble-setup

Cheong2K, 2017-09-28:
Thanks @TechplexEngineer, it is true.

closed 2017-09-28 by Cheong2K