redbear/Duo #32: SoftAP not wornking after flash app

bitShakeMX, 2017-07-13:
I have 10 redbear duo, all received the 0.3.1 update. After that, when I get into listen mode I can't see the SSID on my wifi (PC, iPhone ...etc)

I made a hard reset, SoftAP back to work. Repete all the process, set up with the credentials, flash the app via Web IDE, get into listen mode and stoped working again, even with example Blink app

What I can do?

Cheong2K, 2017-07-18:
Did you follow this?
https://github.com/redbear/Duo/blob/master/docs/out_of_box_experience.md#configure-wifi-credential

hugomosh, 2017-07-25:
The problem is, that in Listening Mode (the blue flashing), after the uploading an app to the Readbear. It won't display the SSID. 
So if the Readbear is empty it appears the SSID. DUO-xxxxx but when it has a program uploaded, and enable Listening Mode. Nothing shows up.