redbear/Duo #35: RedBear Duo App crashes with iOS v11

Cheong2K, 2017-09-28:
On iOS version 11, using the RedBear Duo App, when I try associating the Duo to a Wi-Fi Access Point/Router, the App crashed.

It works fine on iOS 10.

sparklingsparklers, 2018-02-22:
Doesn’t work for me on iOS 11.2.6 
The app just crashes.

saamerm, 2018-08-14:
I'm not that adept in Swift (just C#) but I tried to clone the project and see where the exception occurs, and I got 

"Thread 1: Fatal error: Unexpectedly found nil while unwrapping an Optional value"

around line 666 (number may be greater than actual because of my comments) of the DuoWifiProvisioningViewController file at this call "alert.addAction(self.passAlertAction!)" 

The self.passAlertAction is null and I tried to follow https://stackoverflow.com/questions/32170456/what-does-fatal-error-unexpectedly-found-nil-while-unwrapping-an-optional-valu but I have no idea what he's saying

Not a 100% sure what I can do about it, any ideas @Cheong2K @OhHeyAlan @XuGuohui or @ChiHung ?
This issue has made my RedBear Duo useless 😿

Danappelxx, 2018-08-14:
It looks like the issue is that `self.passAlertAction` is weak but is [assigned to directly from allocation](https://github.com/redbear/Duo/blob/master/apps/ios/DuoApp/DuoApp/DuoWifiProvisioningViewController.swift#L652). As such it is immediately deallocated and afterwards crashes when unwrapped, at least some of the time (undefined behavior unfortunately). The fix for this is straightforward - simply allocate it on the stack and then pass ownership to the alert. I'll submit a pull request.

Danappelxx, 2018-08-14:
Made the pull request. Let's hope it gets merged! @saamerm I would appreciate if you could test it to make sure my theory is correct.

saamerm, 2018-08-14:
Verified! @Danappelxx 's PR fixes the crash.
To test it/use a working copy: 
1) you need a Mac with xcode, 
2) then clone his Duo repo, switch to the right branch
3) Change the bundle identifier and the provisioning profile to random test stuff, 
4) Connect your phone and Build

wywarren, 2018-08-17:
I cloned the repo but I'm getting an error with the link to Pods_RedBear_Duo.framework . Do I need to build the pods framework separately first? If so how? I do see the pods folder in the iOS project folder but it's just a bunch of nested frameworks inside another project. I suspect the SparkSetup error I'm getting is related to this missing framework?

Thanks for any help you guys can provide. I'm also on iOS 11 and the app store app is crashing on me.

saamerm, 2018-08-17:
@wywarren Hey Warren 
Im not sure why you are getting that error :/
Did you open the workspace and do you have the latest version of Xcode installed?

Danappelxx, 2018-08-17:
@wywarren you need to download cocoapods (`gem install cocoapods`) and run `pod install`, and then everything should build.

wywarren, 2018-08-18:
I ended up opening the DuoApp xcode project and then dragging the Pods excode project file into the DuoApp's project and everything built as expected. I also ran the cocoapods as @Danappelxx suggested. Not sure if that helped it as well. But it got rid of the SparkSetup error as the source code was actually in the project now. Time to continue tinkering. Thanks for the help guys.

wywarren, 2018-08-18:
Just a note: The commenting on DuoBLEProvisionTableViewController.swift should have the filename fixed on line #3 
`//  DuoWifiProvisionTableViewController.swift`
This caused a bit of confusion for me thinking I was editing the wrong file at one point.

But yes after updating the Wifi and BLE swift files everything seems to build and not crash now.